esynr3z/corsair #8: AXI BResp and RResp

stridge-cruxml, 2022-03-15:
Instead of always returning [ok](https://github.com/esynr3z/corsair/blob/master/corsair/templates/axil2lb_verilog.j2#L77), if trying to write to a read only address, or read a write only, or have a bad address return an error instead.

stridge-cruxml, 2022-03-27:
all good. I might have a go at some point. But it sounds like you desire this feature to be optional? Add a parameter to verilog generation class with something like `enable_axi_response`?

iDoka, 2022-03-31:
+1
I think verilog parameter or define is reasonable solution for it.

arnfol, 2023-03-16:
While I am not expecting this issue would be resolved soon-ish, while digging templates I found a possible workaround. This feature is not well-documented, but for Verilog and VHDL generators you can define `read_filler` in the config file. Setting it to some meaningful value like `0xDEADC0DE` will give you the opportunity to debug if you are trying to read a bad address.

arnfol, 2023-03-16:
I was wrong about not being well-documented: [here](https://corsair.readthedocs.io/en/latest/config.html#verilog)

esynr3z, 2024-10-08:
It's been a while...

I plan to add some knobs to HDL generator classes to enable error propagation.

This is not the nearest future, but is going to be done eventually.