esynr3z/corsair #12: CMSIS SVD

raffi-g, 2022-04-04:
Hi! First of all, thanks for the great work so far!

I would like to add a generator for CMSIS SVD files (a register description format for debugging, see [here](https://www.keil.com/pack/doc/cmsis/SVD/html/svd_Format_pg.html)).

I've already done most of the work. The only thing is, that SVD wants to know, in which peripherals the registers are, e.g.

    <peripherals>
        <peripheral>
            
            <registers>
                
            </registers>
        </peripheral>
        <peripheral>
            
        </peripheral>
    </peripherals>

At the moment this information is not provided in the register map. In order to minimize the impact on the current defintion and also be backwards compatible, I would suggest to just add an entry "peripheral" to each register, like so:

    {
            "name": "CSR_NAME",
            "description": "",
            "address": "0x20",
            "peripheral": "my_peripheral",
            "bitfields": []
    }

What's your opinion? Or is this not wished at all? Then I would just leave it on my fork.

esynr3z, 2022-04-22:
Hello! 

Of course I'm interested in SVD generator! CMSIS SVD was one of the sources of inspiration for me.

I see that you've done plenty of work already. And I'm planning to do a big refactoring, so I'm sure there will be much elegant way to integrate support of several peripherals. For example, to add several register maps to `csrconfig`. I think it is much convenient to place register map of each peripheral in a separate file.

raffi-g, 2022-05-10:
Hi
Goot to hear. If I can support you in any way, just let me know.

esynr3z, 2024-10-08:
This is nice feature that has to be done, eventually. However, for me now refactoring and 2.0 is the milestone, so this feature take place only after that.

Moreover, I have plans to add support of multiple register maps in one file, so SVD will get "peripheral" out of the box.