esynr3z/corsair #15: #enhancement: wavedrom bitfield adding `lanes` to csrconfig

iDoka, 2022-04-04:
At this moment if I correct understand the `lanes` for wavedrom bitfield creation sliced for 16 bits per lane:
https://github.com/esynr3z/corsair/blob/master/corsair/generators.py#L104

In generic case it is good solution but in some specific cases (almost all) would be nice manually specify number of `lanes` in  `csrconfig`.

Just an example: my most adorable documentation - any datasheet from Atmel company. They use config 8 bits per lane and it is convenient way to have enough volume to print field name (even if it's not the shortest name):

[image]

Would be nice to have such flexibility in future release 🙏🏻

arnfol, 2023-03-16:
I accidentally found out that now corsair always generates one-line registers, which is not an expected behavior. Maybe [wavedrompy](https://github.com/wallento/wavedrompy) API changed

esynr3z, 2024-10-11:
I agree that that would be a nice knob for documentation generation.