esynr3z/corsair #23: Move access and hardware from bitfields to reg

stridge-cruxml, 2022-07-29:
This would be a big change, but I think it would make more sense to have the `hardware` and `access` attributes at the `reg` level.

I can't think of a reasonable use case to have a single register 24 bit with 8 bits a read only status value and another set of 16 bits as a read/write queue for example and should be explicitly disallowed requiring separate registers.

iDoka, 2022-07-29:
I'm totally disagree to moving it to register level.

Just see any Technical Reference Manual for modern uC. Furthermore, this happens more often than you can imagine (if I was paid one Satoshi 💰 for each peripheral module where such mixing is needed, then I would not be talking to you engineers here 😎).

If your goal to getting compact form of initial description of regs I can suggest to add the `hardware` and `access` attributes at the `reg` level and at the same time saving ability to overriding the `hardware` and `access` attributes (if it defined here) at the `bitfields` level. Actually, the `hardware` and `access` attributes at the `reg` level will be define a "default values" for bitfields of this reg.

stridge-cruxml, 2022-08-01:
It was more when I was modifying the templates, having it at the bitfield level complicates changes.

I suppose you could have a fifo and a size count in the example above. Still seems messy to me though, but I understand where you are coming from.

closed 2022-08-01 by stridge-cruxml