esynr3z/corsair #31: base_address for RTL modules

vborchsh, 2022-12-05:
Hello everybody,

Maybe I didn't get the point, but I see that base_address variable is used nowhere but in the C/Python headers and so on. Shouldn't the variable used also for RTL modules for absolute address calculation? Something like:

```
absolute_reg_address = base_address + reg_address
```

Example:

Corsair register
```
-   name: UID
    description: Device ID register
    address: 0
    bitfields:
    -   name: UID
        reset: 0xAABBCC01
        width: 32
        lsb: 0
        access: ro
        hardware: i
        enums: []
```

I connect corsair registers map to the AXI interconnect core. The address of the core in the AXI interconnect, let's say, 0x8000_0000. It is a "base address" of the core. And this case corsair's core doesn't accept input addresses **0x8xxx_xxxx** becasue of **0x8**. Corsair waits an address without base offset.

Thanks.

PS. For sure, there is a workaround with select only part of input address for Corsair, but still

arnfol, 2023-03-06:
Hi, @vborchsh!

Thank you for pointing out the problem. I think that part-selecting address is, actually, not a workaround, but the right way to use registers.

The problem with setting "base address" or "absolute address" in the hdl code is that the code becomes not really portable. In general, you can use generated registers in an IP core, which could be instantiated several times in a project or across different projects. In my view, generating the absolute address of a register map should be a job of top-level design, rather than corsair.

But I see that in C/Python there is `CSR_BASE_ADDR` defined, which is confusing.

arnfol, 2023-03-16:
I am closing this issue for now, feel free to reopen it, if I have not answered some of your questions

closed 2023-03-16 by arnfol