esynr3z/corsair #50: feature request: support symbolic constants for `base_address`, (do not require it to be a number)

v0lker, 2023-06-15:
i think it would be nice to be able to write this in the `config`
```
[globcfg]
base_address = MY_SYMBOLIC_CONSTANT
```

or maybe even
```
[globcfg]
base_address = MY_SYMBOLIC_CONSTANT + SOME_OFFSET
```

and to get this in the C headers:

```
#define REGS_BASE_ADDR MY_SYMBOLIC_CONSTANT
```

or

```
#define REGS_BASE_ADDR (MY_SYMBOLIC_CONSTANT + SOME_OFFSET)
```

respectively. maybe we don't need to be able to handle arbitrary strings, but it's probably the same amount of effort (?)

does anybody else like the idea?

v0lker, 2023-06-15:
implementation could look like this: https://github.com/esynr3z/corsair/pull/51

esynr3z, 2024-10-11:
It's a nice feature, really. But it opens a portal to symbolic calculations instead of concrete numbers, which is not an easy thing to implement and maintain within all templates.

I've also answered to the quite similar topic [here](https://github.com/esynr3z/corsair/issues/49#issuecomment-2408075945), if you are interested.

I thinks this won't be implemented soon, but, indeed, this has to be thoroughly thought.