esynr3z/corsair #61: #enhancement : add wishbone b3

stdefeber, 2024-06-26:
Although very similar to Avalon, would it be possible to add a Wishbone B3 compliant interface ?

stdefeber, 2024-07-19:
I have been working on this issue myself.
Although generation seems to be fine I cannot verify the changes since there is some dependency on Win32 packages ?

martijnbastiaan, 2024-07-21:
I don't see any Windows-specific dependencies in `setup.py`. Could you elaborate @stdefeber? What commands are you issuing? What errors do you see?

stdefeber, 2024-07-22:
Command issued:

`python3 -m pytest -v -n auto`

I am running corsair in a virtual environment.

The errors I get, during the tests, are:

```
[gw2] [ 32%] FAILED lib/python3.12/site-packages/pip/_vendor/colorama/tests/ansitowin32_test.py::AnsiToWin32Test::test_native_windows_ansi 
lib/python3.12/site-packages/pip/_vendor/colorama/tests/ansitowin32_test.py::AnsiToWin32Test::test_osc_codes 
[gw2] [ 32%] FAILED lib/python3.12/site-packages/pip/_vendor/colorama/tests/ansitowin32_test.py::AnsiToWin32Test::test_osc_codes 
lib/python3.12/site-packages/pip/_vendor/colorama/tests/ansitowin32_test.py::AnsiToWin32Test::test_reset_all_shouldnt_raise_on_closed_orig_stdout 
```

and while gathering final info, just to grep 1:

```lib/python3.12/site-packages/pip/_vendor/colorama/winterm.py:31: AttributeError
__________________________________________________________ WinTermTest.testSetConsole __________________________________________________________
[gw1] linux -- Python 3.12.3 /home/stdefeber/projects/corsair/.venv/bin/python3

self = <pip._vendor.colorama.tests.winterm_test.WinTermTest testMethod=testSetConsole>
mockWin32 = <MagicMock name='win32' id='128564182793776'>

    @patch('colorama.winterm.win32')
    def testSetConsole(self, mockWin32):
        mockAttr = Mock()
        mockAttr.wAttributes = 0
        mockWin32.GetConsoleScreenBufferInfo.return_value = mockAttr
>       term = WinTerm()

lib/python3.12/site-packages/pip/_vendor/colorama/tests/winterm_test.py:104: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <pip._vendor.colorama.winterm.WinTerm object at 0x74edae743620>

    def __init__(self):
>       self._default = win32.GetConsoleScreenBufferInfo(win32.STDOUT).wAttributes
E       AttributeError: module 'pip._vendor.colorama.win32' has no attribute 'GetConsoleScreenBufferInfo'

lib/python3.12/site-packages/pip/_vendor/colorama/winterm.py:31: AttributeError

```