esynr3z/corsair #69: Expand Project Maintainership/Ownership

iamjjw, 2024-10-02:
Hi @esynr3z !

I think I can speak for many if I say that I absolutely love this project. I use it constantly during work and we have expanded a fork of this repo to include some more things, such as GUI generation for Python, expand the selection of busses to connect, etc.

Seeing that the last approved pull request is from April of 2023, I was wondering about the future of this project. It is clear to me that interests, priorities and much more can change over time, and I would not fault you for it. If this is the case, would you (or whoever is maintaining this project) consider including other maintainers to this project to keep it alive?

Either way, I hope that you are well, and greetings from the team at IMES at OST in Eastern Switzerland.

esynr3z, 2024-10-03:
Hi @iamjjw !

You've raised interesting question.

First of all, I'd like to thank you for your kind words, I'm really happy to hear that the project is actively being used and makes someone's life easier. I'm also quite curious - why Corsair? Why not SystemRDL or other alternative?

You are right that my interests shifted away a bit, and this multiplied by severe lack of free time gives us what you can see now.
I've been pondering for a while trying to answer the same question you raised to myself and still have no single answer.

- Archive the project? No, it indeed sounds not right while project is being actively used.
- Completely transfer ownership? Not sure. I've put a lot of effort and passion into it and it is quite hard to leave it like that.
- Include more maintainers? Sounds like a right solution. However, without "being an owner" (making reviews, fixing things, creating a roadmap and planning the whole project development goals) it is more like the second option.

It is been a while since this project has started and now I definitely gain more experience and become better at Python, automation and even planning and delegating things. I see now that a lot can be rewritten and organized in a better way, and I even have tried to create such a roadmap to reorganize this project. I also see amount of issues and pull requests, and amount of technical debt. And it is a shame that I ignored them most of the time.

So I have everything to make this project even better. Except one thing - motivation.
Now I see that in medium to big companies people prefer SystemRDL with custom rendering templates.
So I also started thinking maybe some standardized thing is better that inventing CSR description format one more time?

Why Corsair is important for you? Why it can be important in general?

P.s. sorry for switching to a bit different topic, but let's discuss all that here anyway, it can help me to make a decision and maybe can lead to a new page for Corsair itself.

iamjjw, 2024-10-04:
What ohofherr is saying is essentially what it boils down to in my use case. Being an educational facility, many of our users are students. There is a lot of demand in education for low-barrier-of-entry solutions to common problems, and most of the time, I have to write them myself.

Corsair has provided me with a solution for this specific case. I could also imagine someone wanting to do a small one-off project but has never used a CSR tool before; it's a lot easier to start with Corsair.

esynr3z, 2024-10-04:
Thank you! Your points make sense. 

Now I usually surrounded with dirty enterprise tools and processes, so I started to forget how simplicity can be valuable and important.

@ohofherr 
SystemRDL is exactly the beast for enterprise. Before even use it you have to invest effort and time into comprehending specification and the language itself. However, it is quite powerful and can solve magnitude higher CSR cases than Corsair.

iamjjw, 2024-10-14:
So I think that we agree on the fact that corsair needs to stay! 😄 
My question is how this can be sustainable for you. For example, would you be comfortable with "just" looking at pull requests? In other words, whoever want's an issue fixed can do so, but you would be the final arbiter of it being contributing to the project?

esynr3z, 2024-10-14:
Well, I have decided to resume work on this project and improve it consistently.

Right now, I realize that it won’t be enough for me to just 'watch the pull requests.' I’ve outlined a roadmap and would like to follow it.

In short, I want to refactor the internal code of Corsair to make it stricter, more stable, and easier to add new features. I also want to improve automation a lot, so changes and fixes become more predictable. The user API will stay mostly the same, as I think it works well, but unfortunately, some breaking changes will be necessary.

In fact, I’ve already processed the backlog and started working on version 2.0.

Later, I suppose we can follow commonly used flow to make issues fixed:

- discuss an issue
- find an assignee
- get a PR
- PR is reviewed
- CI is green
- PR is merged
- issue is closed

iamjjw, 2024-10-14:
Great to hear. I'll be happy to watch the progress. Thank you very much!

esynr3z, 2024-10-15:
Thank you for this issue. Some motivation and inspiration is what I actually needed.

closed 2024-10-15 by esynr3z