esynr3z/corsair #89: support bram/fifo interface

roybod, 2024-10-26:
we
re
addr
wdata
rdata

esynr3z, 2024-10-27:
Hi!
FIFO interface is already supported and can be enabled for a field via `f` [hardware flag](https://corsair.readthedocs.io/en/latest/regmap.html#bit-field). You could also check result in [examples](https://github.com/esynr3z/corsair/blob/14dc5d40a52f7efac07393d0a53623fc231f8f77/examples/regmap_yaml/hw/regs.v#L12). However, interface is a bit different from suggested above - there's no address and FIFO can apply backpressure with write-ready and read-valid signals.

As for BRAM being mapped into register map - there is already issue for this: #39. This is planned for future versions.

roybod, 2024-10-27:
thanks.
i came across your tool because i used AIRHDL and found out that the developer died...

esynr3z, 2024-10-27:
I'm sorry to hear that. Is that tool now unmaintained?

Maybe, if you have some free minute, could you please share some feedback about airhdl? Maybe what was the best and worst parts of it? This could help Corsair in the future.

roybod, 2024-10-27:
there is a working website airhdl.com
it is still working
you can create a free user and test one register map